budude2/openfpga-GBC #1: RTC Done Broke

Xukkorz, 2024-01-03:
When Adding RTC data to save with no RTC (Pokemon Gen 2), time appears to move forward very quickly, causing flashing. Appears to stop eventually.

Posting issue per Request.

insanedude63, 2024-01-03:
Expanding this issue
subsequent loads of the same save result in time having to be set again

better than time standing still between saving and loading though

MaciekBaron, 2024-01-03:
This sounds like fast forward being triggered accidentally.

BreakPoints, 2024-01-03:
> This sounds like fast forward being triggered accidentally.

I don't belive so, I was walking around the overworld in Pokemon at a normal speed but the time of day was flickering between states. Walking and standing also seemed to affect the speed at which the flickering was happening.

MaciekBaron, 2024-01-03:
Thanks for providing more details, that makes sense!

Xukkorz, 2024-01-03:
Its not fast forward. Budude is aware of the issue he just asked me to put the issue in.

random11x, 2024-01-04:
The RTC is initialized to 0 for save files that dont have RTC set. If you load a game it will increment the RTC in a loop until it catches up to whatever initial RTC time is being set. If your pokemon save says "Time Not Set" press "Continue" and make note of the date and time it suggest. Back out and repeat. It should say a completely different time if its still in this state, until the RTC catches up. If you leave the game open for about 2 minutes on the save select screen it should get to a point where you can do this a few times and the suggested time doesnt change. It is then safe to set the time and when you get in game it should be set correctly without the flickering between day and night. In my experience as long as you dont try and open the save in the spiritualized core afterwards, it should stay set.

budude2, 2024-01-05:
This should be fixed in v1.0.4!

ninbura, 2024-01-10:
Issue still open, so I assume it's still known to be an issue. But this still occurs in version 1.0.5.