budude2/openfpga-GBC #4: Support for Cartridge Peripherials (eg: Rumble, Accelerometer, Camera)

HeyItsLollie, 2024-01-03:
There's a good handful of Game Boy / Color titles that had additional features built into the cart itself, eg: [rumble](https://www.reddit.com/r/gamecollecting/comments/bvgeax/every_gameboy_color_rumble_cart_released_i/?utm_medium=android_app&utm_source=share), [accelerometer](https://gbhwdb.gekkio.fi/cartridges/mbc7.html), the Game Boy Camera. It'd be great to be able to utilize physical game carts or flashcarts ([[1]](https://www.ezflash.cn/product/ez-flash-omega-definitive-edition/), [[2]](https://shop.insidegadgets.com/product/gba-32mb-256kbit-fram-save-flash-cart/), [[3]](https://shop.insidegadgets.com/product/gba-32mb-256kbit-fram-gyrorumble-flash-cart/)) equipped with these features, and would really elevate this core.

In the case of Game Boy Camera:
There's a couple rare alternative versions of GB Camera (The ["Gold"](https://niwanetwork.org/wiki/The_Legend_of_Zelda:_Ocarina_of_Time_Game_Boy_Camera) Ocarina of Time version, the unreleased & leaked [Hello Kitty Pocket Camera](https://niwanetwork.org/wiki/Hello_Kitty_Pocket_Camera)), along with a couple homebrew replacements ([Photo!](https://github.com/untoxa/gb-photo), the discontinued-but-still-used [PXLR Studio](https://github.com/HerrZatacke/2bit-pxlr-studio)) that require a custom flashcart and parts harvested from a physical GB Camera. Spiritualized1997's [GBA](https://github.com/spiritualized1997/openFPGA-GBA) and [SGB](https://github.com/spiritualized1997/openFPGA-Super-GB) cores have shown it's possible to utilize cart features / load entire carts alongside the core, so if it's possible to utilize a physical unmodified GB Camera cartridge, it'd make these alternative versions infinitely more accessible.

MykeLiaz, 2024-01-04:
Wait so they core from spiritualized of the gba can read real cartridges?

HeyItsLollie, 2024-01-04:
Spiritualized's _SGB_ core can run real cartridges. Their GBA core can use the rumble feature from real rumble carts.

budude2, 2024-01-05:
Rumble support has been added. I am not sure about the other two peripherals yet.

HeyItsLollie, 2024-01-06:
Wicked, thank you for looking into it!

budude2, 2024-01-06:
Looks like I didn’t do it right, so rumble is super weak