budude2/openfpga-GBC #6: desaturated colours

Noble3781, 2024-01-03:
Compared to other GBC Cores the colours are desaturated in the no filter mode.

aggiemex, 2024-01-04:
> Compared to other GBC Cores the colours are desaturated in the no filter mode.

This also happens on the GB core. When using the Original DMG filter on this core the colors don't match the cartridge. 

See attached photo of the mushroom in Super Mario Land. It should have a dark top but using this core it looks white.

[image]

pyth27, 2024-01-04:
@aggiemex The game is running in GBC mode with the DMG filter applied. Looks fine to me in the GB core from the latest release (1.0.1).

aggiemex, 2024-01-04:
@pyth27 Thank you! Updating to 1.0.1 and using the GB core resolved the issue!!!

budude2, 2024-01-04:
Desaturated colors in GBC core should be fixed in 1.0.2

closed 2024-01-04 by budude2