budude2/openfpga-GBC #11: Screen Tearing Occurs Following FastForward Usage

Psyclaire, 2024-01-04:
Seemingly after about a minute of using fast forward, a single horizontal line of torn pixels appears towards the top of the display. This is remedied by reloading the core.

jz-k, 2024-01-14:
The tear seems to manifest on a different line between fast-forwards. I've been able to clear it from the screen by fast-forwarding briefly after it appears. Sometimes it will move, but often it will go away completely (only to return when I use the FF feature again later).

Psyclaire, 2024-01-14:
> The tear seems to manifest on a different line between fast-forwards. I've been able to clear it from the screen by fast-forwarding briefly after it appears. Sometimes it will move, but often it will go away completely (only to return when I use the FF feature again later).

This is the case too, it appears the frame can “land” on a tear, between fast forward cycles.

Rahkeesh2000, 2024-08-26:
This issue also seems to affect the new save states in 1.3.1. Even creating a save state (not even loading) with FF off is enough to trigger the tearing.