budude2/openfpga-GBC #12: Feature Request: Option to Toggle FastForward

Psyclaire, 2024-01-04:
Option to *Toggle* FastForward with the mapped button as opposed to holding the mapped button.

random11x, 2024-01-07:
double tap mapped button fast. Wait a beat then tap it one more time. FF will stay on.

Psyclaire, 2024-01-07:
> If you press and release the button quick enough, FF will stay on. Tap it as if you were tapping a pot to check if it was hot.

I’ll have to try this. Either way an easier method would be nice. I’d be curious to know if the current method is intentional or not.

random11x, 2024-01-07:
> Either way an easier method would be nice. I’d be curious to know if the current method is intentional or not.

Its not intended lol. But if you try it... its pretty easy.

danoisthemano, 2024-01-10:
> If you press and release the button quick enough, FF will stay on. Tap it as if you were tapping a pot to check if it was hot.

Can confirm that this exploit(?) works for me as well.

Psyclaire, 2024-01-10:
> Option to _Toggle_ FastForward with the mapped button as opposed to holding the mapped button.

Commenting to update my original issue. As @random11x mentioned you can essentially force the FastFoward button to act as a toggle by _very_ quickly flicking the button. I suppose this works for now, but it would certainly be nice if there was an option in the cores settings to allow for either _Trigger or Toggle_.

closed 2024-08-20 by budude2