budude2/openfpga-GBC #15: Feature request: real cartridges

MykeLiaz, 2024-01-04:
Can you make it read Gameboy and Gameboy Color cartridges so I can give it some use with a core?

reminon, 2024-01-04:
This would be interesting in conjunction with flaahcarts that don't have rtc.

MykeLiaz, 2024-01-04:
> This would be interesting in conjunction with flashcarts that don't have rtc.

Yea i know it would be interesting to use a real cartridge and give ir a use with the rtc or much more

ninbura, 2024-09-15:
This would also allow for save extraction from retail cartridges without a cart dumper. 
1. Load a retail cart
2. load the in-game save
3. create a save state
4. load the same game via rom
5. load the save state you created with the cartridge
6. save in-game
7. extract the save data from the Micro SD card

Or it may even be possible to direct the core to save to the cart & the Micro SD in the Pocket?