budude2/openfpga-GBC #7: Feature request: Support for custom palettes

nortakales, 2024-01-04:
https://www.analogue.co/developer/docs/custom-palettes

nortakales, 2024-04-03:
So, it looks like custom palettes were added at some point, which is freakin awesome! Thanks!

Is there any way for us to get custom palettes AND grid lines though?

When applying a custom palette and any of the display modes that have grid lines, the results do not mix well, as reported in https://github.com/budude2/openfpga-GBC/issues/47

sjkorvin, 2024-04-13:
I noticed this issue today. I do not know the meaning of the term "byteswapped," but none of the palettes downloaded through pupdate looked correct after I changed their file extensions from .pal to .gbp.

nortakales, 2024-04-14:
I built an editor for `.gbp` files here: https://www.nortakales.com/page/gbpEditor

For whatever reason I could not find one anywhere on the internet 🤷‍♂️

ButterCCM, 2024-04-25:
So the core isn't able to use any of the .pal files? I hope that gets changed I don't want to have to convert them all.

Xbudz, 2024-04-26:
> I built an editor for `.gbp` files here: https://www.nortakales.com/page/gbpEditor
> 
> For whatever reason I could not find one anywhere on the internet 🤷‍♂️

Thanks for this. Any chance at allowing batch conversion?

nortakales, 2024-04-26:
> > I built an editor for `.gbp` files here: https://www.nortakales.com/page/gbpEditor
> 
> > 
> 
> > For whatever reason I could not find one anywhere on the internet 🤷‍♂️
> 
> 
> 
> Thanks for this. Any chance at allowing batch conversion?

Good idea. Give me a week or so, I should be able to find the time to do that.

Xbudz, 2024-04-26:
> > > I built an editor for `.gbp` files here: https://www.nortakales.com/page/gbpEditor
> > 
> > 
> > > 
> > 
> > 
> > > For whatever reason I could not find one anywhere on the internet 🤷‍♂️
> > 
> > 
> > Thanks for this. Any chance at allowing batch conversion?
> 
> Good idea. Give me a week or so, I should be able to find the time to do that.

You're awesome, thanks for your efforts.

nortakales, 2024-04-29:
> You're awesome, thanks for your efforts.

Batch conversion tool is complete, just scroll down and find the "Batch Conversion" section: https://www.nortakales.com/page/gbpEditor

Xbudz, 2024-04-29:
> Batch conversion tool is complete, just scroll down and find the "Batch Conversion" section: https://www.nortakales.com/page/gbpEditor

On behalf of everyone, thank you sooooo much.
This will really bring a bunch of smiles to folks.
We appreciate you so much.

closed 2024-08-21 by budude2