budude2/openfpga-GBC #8: GB core doesn't start.

k5jat, 2024-01-04:
I copied the Cores and Platforms directories to my sd card. When I try to set your core to run roms, it goes to another directory listing, like a crash on boot. I set it to the other core, and it runs fine.

pyth27, 2024-01-04:
Do you have a gb_bios.bin in your `/Assets/gb/common` folder?

aggiemex, 2024-01-04:
@pyth27 I don't know if this is the right place to ask, but using the Spiritualized1997 GB/GBC core I am able to Sleep/Wake the Analogue Pocket while playing in OpenFPGA, but not with this core. Is that because Sleep/Wake requires Save States to be implemented? If so I will be patient since Badude2 noted that's on the roadmap and is a WIP. :) :)

pyth27, 2024-01-04:
@aggiemex Correct, the sleep functionality requires save states to work.

aggiemex, 2024-01-04:
@pyth27 Thanks again!! +1000 internets for your helpfulness! :o)

k5jat, 2024-01-04:
> Do you have a gb_bios.bin in your `/Assets/gb/common` folder?

It doesn't go in that directory. It goes in `/Platforms/_images` and it should be gb.bin, like in the zip file.

pyth27, 2024-01-04:
Check the [readme](https://github.com/budude2/openfgpa-GBC/blob/master/README.md), you need a BIOS file with the correct name in the correct place: 

> Place the GBC bios in `/Assets/gbc/common` named "gbc_bios.bin", and the GB bios in `/Assets/gb/common` named "gb_bios.bin".

k5jat, 2024-01-04:
> Check the [readme](https://github.com/budude2/openfgpa-GBC/blob/master/README.md), you need a BIOS file with the correct name in the correct place:
> 
> > Place the GBC bios in `/Assets/gbc/common` named "gbc_bios.bin", and the GB bios in `/Assets/gb/common` named "gb_bios.bin".

None of that makes sense, but I'll try it.

pyth27, 2024-01-04:
Hint: you'll need to acquire these bios files from somewhere else, they cannot be shared here because of copyright reasons. 

If that doesn't help I suggest you use tools like [pocket updater](https://github.com/mattpannella/pocket-updater-utility) or [Pocket Sync](https://github.com/neil-morrison44/pocket-sync) which install the core (and bios files) for you.

k5jat, 2024-01-04:
I got the right file in the right spot. Spritualized used the file name `dmg_bios.bin.` I found a `gb_bios.bin` in my gbc assets directory so I copied it over to the gb assets directory.

Anyway, it works now. Hopefully, the updaters get updated soon so they can at least install it correctly. Thanks for your help!

closed 2024-01-06 by budude2