budude2/openfpga-GBC #9: Feature request: Sleep and Save States

nortakales, 2024-01-04:
I know this is already on your list of things you're working on, but I figured we could open an issue for it so people have a place to +1 rather than cluttering other issues

SaulGoodman1337, 2024-01-04:
Yup, this is urgently needed to make this wonderful core a daily companion. Thank you for this and I hope that Sleep&Save is a high priority in your todo.

Is there somewhere to donate for you?

budude2, 2024-01-06:
You can donate at https://ko-fi.com/budude2 if you so desire.

ninbura, 2024-01-10:
Is the planned implementation simply to deal with offsets after state change based on the old in-game time? Or to actually peg in-game time to real time?

It'd be most preferable to have some sort of toggle that syncs in-game time with the day/time on the Pocket.

SaulGoodman1337, 2024-01-11:
@budude2 
And are you making progress with the SaveState Branch? I'm trying to understand what you're doing but failing :D

ninbura, 2024-01-17:
> Is the planned implementation simply to deal with offsets after state change based on the old in-game time? Or to actually peg in-game time to real time?
> 
> It'd be most preferable to have some sort of toggle that syncs in-game time with the day/time on the Pocket.
IDK why, but I just immediately assumed that this was about correcting RTC de-sync with the use of sleep/save states. Now I see that this is simply about the implementation of sleep/save states in general.

Is there any plans to account for RTC de-sync once sleep/save states are implemented? For example, with an EZ-Flash Omega DE flash cart, RTC/in-game time is adjusted to match the date/time set on the cart every time a state is loaded. Without such functionality, I foresee Pokemon players avoiding usage of sleep/save states once implemented. It's fairly tedious to manually reset RTC data / readjust the clock via the in-game exploit.

SaulGoodman1337, 2024-03-10:
@budude2 any News on this? :(

budude2, 2024-03-10:
> @budude2 any News on this? :(

Now that I have fixed the save file handling I can turn my attention back to getting save states working

SaulGoodman1337, 2024-05-26:
We need you...

CraneInForest, 2024-08-03:
@budude2  any News on this? We're looking forward for this~~~

budude2, 2024-08-03:
I’ve been working on it. Unfortunately it doesn’t seem as easy as just hooking up the mister parts and using it. So I am going to have to rewrite portions of it. I needed to make some changes to make the cart module use external ram so I had more ram for debugging. As of now, the games just crash when I try to restore save states. Hopefully I can get it working soon

budude2, 2024-08-20:
Added in the newest version

closed 2024-08-20 by budude2

SaulGoodman1337, 2024-08-20:
> Added in the newest version

feel loved ! Thanks for this feature, I'll try it out tomorrow and enjoy a few classics again.

CraneInForest, 2024-08-21:
> Added in the newest version

I tried it with excitement just now. It works extremely well. Thank you for your hard work! I truly appreciate you!!

SaulGoodman1337, 2024-08-21:
Yep, it works like a charm! Many thanks