budude2/openfpga-GBC #19: Issue: Controlls not working in Alleyway (GB)

ThreeM-MMM, 2024-01-05:
When attempting to play Alleyway on the Gameboy Core, the D-Pad doesn't work.

## Expected Behavior
D-Pad controls the Charakter

## Current Behavior
Character does not move

## Possible Solution
When using the spiritualized core, the controls function as expected.

## Steps to Reproduce
1. Start Analogue Pocket
2. Select openFPGA and Gameboy
3. Change core to budude2 GB core
4. Select Alleyway rom
5. Push Start to start the game
6. Try to move the charakter

## Context (Environment)
AnalogueOS: v2.0
Core version: v1.0.3
Rom CRC/Sha1: 5cc01586 (validated)

SaulGoodman1337, 2024-01-05:
I can confirm that.

nortakales, 2024-01-06:
The paddle immediately moves to the left of the screen, as if D-Pad Left is being held down (hence no other directions working). Hopefully that gives a clue to what is wrong.

MP2E, 2024-01-07:
Just tested the latest MiSTer GameBoy Core this is ported from and the issue is not present there

Dahak66, 2024-01-08:
Alleyway controls are working in 1.0.5.

Dahak66, 2024-01-08:
Alleyway controls are working in release 1.0.5.

MP2E, 2024-01-08:
Yep confirmed! It's working perfectly now

ThreeM-MMM, 2024-01-08:
Confirmed. Working now. Thanks

closed 2024-01-08 by ThreeM-MMM