budude2/openfpga-GBC #23: Error in framework File ID [18] too large with LSDj

sdhizumi, 2024-01-08:
Description:
- I saw an error message of "Error in framework File ID [18] too large" while I try to boot LSDj v9.3.9 with savefile that bring from external emulator.
- I could boot up LSDj v9.3.9 with no savedata. This core is generating savadata when I boot this at first time, but its not same intial savefile that was booted on actual GB/GBC or emulators (like SameBoy).

I want to:
- Load savefiles that is come from actual cartridge or emulators.
- Save savefiles that is bring to actual cartridge or emulators.

Probable cause of this problem:
- LSDj's savefile is 128kilobytes, but this core generated 33kilobytes savefile (It seems like 32kilobytes with something footer).
- This core can't load full size (raw) save file that came from actual cartridge or emulators now.

Unknown points for me:
- The savefile which was created by this core seems working as 128kilobytes with LSDj. LSDj' s song screen says this RAM has BF blocks (=max capacity, 128kilobytes). Is that footer has property of save ram?

budude2, 2024-01-17:
I made the poor assumption that 32KB was the largest RAM size for carts. I’m currently working on redoing how the saving and loading works, along with trying to add savestate support.

budude2, 2024-02-11:
Fixed in latest release (v1.0.7)

closed 2024-02-11 by budude2