budude2/openfpga-GBC #24: Feature request: Abilty to disable fast forward

tngliker, 2024-01-08:
No matter which button I assign it to, I always end up accidently Fast Forwarding when I don't want to. It is possible to assign it to no button in the settings, but this doesn't seem to do anything. Would love to be able to just turn it off completely. Thank you!

tngliker, 2024-01-24:
> You can unmap the button in the Analogue Pocket menu bro...

as I wrote in the original post, that doesn't work bro

hallmeier, 2024-02-25:
Yeah, there seems to be a bug that completely disregards the assigned button and always used R for fastforward. I managed to fix (#40) it, but then I noticed there does not seem to be a way to disable a button by the controls menu. Assigning no button lets it be triggered by the default button again. It is possible to assign the L2, L3, R2 or R3 button, that cannot be triggered by the Pocket, by editing the input_persist.json in the Settings folder. But since that is a dirty hack, I also added a dedicated menu option to disable fastforward in #42. For now, you can get the compiled cores with this feature [here](https://github.com/hallmeier/openfgpa-GBC/tree/master/pkg).

Dawbo66, 2024-02-26:
> Yeah, there seems to be a bug that completely disregards the assigned button and always used R for fastforward. I managed to fix (#40) it, but then I noticed there does not seem to be a way to [...]

 Where is the download button ? i hate github menu system.lol

Dawbo66, 2024-02-26:
Filters have always been available haven't they ?

hallmeier, 2024-02-26:
Not with the [Spiritualized core](https://github.com/spiritualized1997/openFPGA-GB-GBC) everyone was using before budude2 released this one.

Dawbo66, 2024-02-26:
> Not with the [Spiritualized core](https://github.com/spiritualized1997/openFPGA-GB-GBC) everyone was using before budude2 released this one.

I know, just sounded like the user couldn't have filters before your fork of the app

Dawbo66, 2024-02-26:
> i couldn't have filters without also having fast forward before this fork of the app, which is what i said.

So on 1.07 you couldn't use filters ?, before this one , that's how I saw it, but I may have read it wrong.

SegaSnatcher, 2024-02-26:
> It's open source, submit a PR?

Its comments like this that I find unhelpful and annoying in the Open Source community. Not everyone is a dev, so the whole "Well its open source, do it yourself" reply I think is quite rude.

hallmeier, 2024-02-26:
> So on 1.07 you couldn't use filters ?, before this one , that's how I saw it, but I may have read it wrong.

No, you always had filters with the budude2 core, but you couldn't disable the fast forward button. You could go without fast forward with the Spiritualized core, but that one had no display filters, which is why they mentioned both things.

budude2, 2024-02-27:
Added in 1.0.8, thanks hallmeier!

closed 2024-02-27 by budude2