budude2/openfpga-GBC #26: Screen cuts witch analogue filters.

hgage2003, 2024-01-09:
When I turn on Analogue GBC filters on GBC core, screen becomes cropped on the left. This issue is only on gbc core, gb core scales fine

DunhamGitHub, 2024-01-13:
> When I turn on Analogue GBC filters on GBC core, screen becomes cropped on the left. This issue is only on gbc core, gb core scales fine

Is it too much to ask to add a screenshot / photo? Common sense, no?

hgage2003, 2024-01-13:
[image]
[image]

SaulGoodman1337, 2024-01-13:
I think your way of communicating and asking for help is really completely wrong. 

I certainly can't recreate it.
You should read through this documentation again and correct your video.json.
https://www.analogue.co/developer/docs/core-definition-files/video-json

hgage2003, 2024-01-13:
I redownloaded the core from this github releases with video.json for testing and nothing changed. My video.json was the same, as in release 1.0.5. Also, I dont think the problem is in video.json - theese screen modes require core responce to work. The same video.json in gb core work perfect.

I think I opened an issue correctly and fully described it, I thought it is common overlooked bug. So I didn't though the screensots was nessesary. But when DunhamGitHub asked for it, I made photos. 
My English is not very good and I did not meen any offence.

SaulGoodman1337, 2024-01-13:
please poste your video.json

Dawbo66, 2024-01-13:
> When I turn on Analogue GBC filters on GBC core, screen becomes cropped on the left. This issue is only on gbc core, gb core scales fine

Try turning integer+ off , with that off I can go through the filters and they work as should.

hgage2003, 2024-01-13:
> > When I turn on Analogue GBC filters on GBC core, screen becomes cropped on the left. This issue is only on gbc core, gb core scales fine
> 
> Try turning integer+ off , with that off I can go through the filters and they work as should.

Oh thank you! This was the issue!
Closed

closed 2024-01-13 by hgage2003

Dawbo66, 2024-01-13:
No worries.. 😁