budude2/openfpga-GBC #28: Pocket Monsters : Crystal Version [Japan] savefile wrong.

Lingline, 2024-01-10:
PM_CRYSTAL | 2048 | 64 | ROM+MBC3+RAM+Battery+Timer
save file 64k not 32k
32k is wrong
selecting girl save game,load game girl change boy
pls fix it

Lycoris2013, 2024-01-11:
This is a peculiar problem with openfpga-GBC, as it works without problem with MiSTer fpga.
It seems to assume only 32 KB+RTC data size when writing.
Changing the data.json file to "size_maximum": "0x16050" will load 64KB of data.

Lingline, 2024-01-11:
I changed the data.json file to "size_maximum": "0x16050" but save date alway 32k.

Xukkorz, 2024-01-11:
The dev is aware of this issue, they didn't realize saves could be bigger than 32k and are working on correcting this.

budude2, 2024-02-11:
Fixed in latest release (v1.0.7)

closed 2024-02-11 by budude2