budude2/openfpga-GBC #30: Error in framework. Core not ready to run

futurepr0n, 2024-01-11:
Hi, I am finding the the GBC core works fine, but the GB one is showing me an error when trying to access same game or file.

Is that normal or should I be able to boot gb from the Gameboy core with no issue? If I load Tetris world with GBC I can get the  DMG palette no issue, but I also wondered if i should be able to load straight from the core as well? Will it conflict if you also have another core installed for gameboy? I had spiritualized core installed and was using that before installing yours along side. I have no issues running the gbc one and selecting your core but the original gameboy one has the error in framework core not ready to run when trying to select gb roms

Dawbo66, 2024-01-11:
Both should run fine, even with other cores on the SD card. Selecting the same game on either core seems to work.

futurepr0n, 2024-01-11:
uninstalled and reinstalled and working fine now, thanks great stuff!

closed 2024-01-11 by futurepr0n

SpeMarX, 2024-01-16:
I’m having the same issue. Did a reinstall, also tried with a new memory card. It’ll work fine for a few days and then throw up the Framework error.