budude2/openfpga-GBC #31: Audio buzzing while playing Final Fantasy Advance

SpiralBrad, 2024-01-13:
Simply play Final Fantasy Advance with this core past the first fake-boss enemy encounter, immediately after name entry.  Have the volume up and you'll hear buzzing.  Issue re: [Sameboy](https://github.com/LIJI32/SameBoy/issues/394), issue re: [Kevtris' core](https://github.com/spiritualized1997/openFPGA-GB-GBC/issues/40) -- it DOES persist throughout the game, in silent parts.

(There is currently an option for "Enable FF Sound") - problem seems to be F6+ still being audible, per [this Sameboy issue comment](https://github.com/LIJI32/SameBoy/issues/394#issuecomment-1250438975) - should there be an option "Enable F6+ Sound"?

hgage2003, 2024-01-15:
Please rename your issue to Final Fantasy Adventure, I thought you mean GBA game Final Fantasy Tactics Advance before visiting your links.

SpiralBrad, 2024-01-15:
> Please rename your issue to Final Fantasy Adventure, I thought you mean GBA game Final Fantasy Tactics Advance before visiting your links.

Done. Huge thanks; brain fart on that one.

clockfort, 2024-03-08:
> (There is currently an option for "Enable FF Sound")

That setting is unrelated, "Enable FF Sound" means "enable sound while using the Fast Forwarding feature"