budude2/openfpga-GBC #34: Add support to run Physical Cart

NeilGSPGUK, 2024-01-19:
NOT AN ISSUE but a Request.

I am very impressed with your cores and thank you very much for all your hard work. I will shortly send you donation but lost the link to that currently.

The Super Game Boy Core from spiritualized https://github.com/spiritualized1997/openFPGA-Super-GB has a run_cartridge.gb file included in the Assets that allows the core to run a physical cart as well as ROM Files from the SD Card.
Would it be possible to implement the same feature in your GB/GBC Core?

NeilGSPGUK, 2024-01-21:
But it doesn't boot the cartridge,  I get the nintendo logo drop down in GB Core or the GAMEBOY Logo in Game Boy GBC Core but the Nintendo part is corrupted. Where as the Main AP Play Cartridge and SGB Core but the carts without issue.

NeilGSPGUK, 2024-01-24:
using the SGB Core there is the run_cartridge.gb this allows the SGB core to run ROMS or using the run_cartridge.gb it will run the Physically connected Cartridge
I moved the run_cartridge.gb to the COMMON folder of the openfpga-gbc and openfpga-gb core folder. and tried to run the Physically Catridge from thes GBC or GBC core, but it doesnt work.

JoshuaRamey, 2024-02-06:
Just commenting to say that I think this feature would be a great addition, and have a similar request based on my particular use case:

First, @825i is wrong and the core cannot load from physical cartridges. The option in the menu that he is referring to is for loading ROM files from the SD card.

As for my specific request, I am not sure if this is even possible but it would be great if we could somehow use this feature to correct the RTC on a physical cartridge. I'm playing Pokemon Crystal and I dumped my save so that I could use a hex editor in order to enable to flag to catch Celebi and then I put that save back on the cartridge. As a result of this, the time set in the game is now incorrect (It's around 30 minutes behind what it should be) and there is no option to change the time in game. I think there might be a tool to edit the time for a save file, but basically any and all fan tools are incompatible with the Japanese version of the game. I also can't use a hex editor because I couldn't find any information online regarding doing so with the Japanese version of the game (Again, the fan community is centered around the English ROMs).

If I am correct in assuming this, the RTC in the core bases the time on something in the hardware similar to how an emulator uses the time set on your PC? If that is the case, perhaps it would somehow be possible to load the game and save data from the cartridge, auto adjust the RTC based on the hardware setting, and then write that data back to the cartridge. If this is at all a possibility (even if the solution is something else that requires multiple steps) I would definitely be interested.

NeilGSPGUK, 2024-02-06:
Your comment isnt here @825i that I just received: You seem to be mistaken. It will run a cartridge provided you set up the .rom file (which points to the cartridge) and the type is the same as the core.

I am CONFUSED!
I have both the budyde2 GB and GBC Cores V1.06
I have a NINTENDO GB Cartridge in the SLOT
I run the CORE and the first menu I see is the ROM SELECTOR
FOR GB I have the run_cartridge.gb ASSETS>GB>COMMON Folder
the same for GBC I have the run_cartridge.gb ASSETS>GBC>COMMON Folder

I can select any ROM from the Folder and it runs the ROM

If I select the run_cartridge.gb with the Physical Cart installed I get:
[image]
I get the same if I do it with the GB or GBC core:
[image]

If I run a ROM image from SD CARD then go back to MENU in the MENU IO have 
DISPLAY MODE
CONTROLS
LOAD CARTRIDGE
RESET CORE
ENABLE RUMBLE
ENABLE FF SOUND
RESET ALL TO DEFAULT
I Select LOAD CARTRIDGE again and select run_cartridge.gb it will show the CORRECT "NINTENDO" Splash screen but then either a corrupt garbled screen or a blank screen

If I Go to the MAIN MENU (NOT OPENFPGA) and select PLAY CARTRIDGE then I get the normal GAME DETAILS LIBRARY PAGE and then I can start the game immediately, so it isn't an issue with the CARTRIDGE.

NeilGSPGUK, 2024-03-08:
Tried without success. I even reloaded cores to a fresh formatted sd card. Then copied the run-catridge.gv from  Spiritualized SBG assets folder. Still just get corrupted nintendo logo as per previous screenshot provided.

Metaru-Shi, 2024-04-24:
Just want to revive this and confirm that it is still not possible to load physical cartridges on the core. A major benefit of this would be to allow fast forwarding on physical cartridges. Please consider adding this feature!
Thanks

haloharry, 2024-05-20:
I also love to see this added.

Hyourinmaru0, 2024-08-21:
I would also love it to see, there is an option in a further release, to hand over and load a inserted physical gameboy cartridge in the GB core and run in SGB mode with all the enhances, not least because the color enhancements for supported games.

chipxsd, 2024-10-19:
you won't be able to just run the `run_cartridge.gb` rom from spiritualized. That rom is nothing but just a message to tell the core to source the data from the cart, since (I think) the early version of OpenFPGA didn't allow any useful menu actions.

The author has to implement this feature into the `openfpga-GBC` core.

also, Github needs an upvote feature.

mechatronics-studio, 2024-11-17:
I've been digging through this core's code for a couple of weeks now. To 100% clear up any confusion, there is currently nothing in the source code that connects up the cartridge pins on the pocket. They are fully disconnected.

I'm in the process now of trying to hook it up. Technically, I've got everything connected (and I'm seeing voltage, clocks, reads, writes, and addresses onto the cartridge pins) but the bios (which I've confirmed is working) isn't talking nicely with the cart.