budude2/openfpga-GBC #38: The RTC on HuC-3 not working v1.0.7

Lycoris2013, 2024-02-14:
The save data size is now handled correctly in v1.0.7. Thanks.
However, I found issues related to saved data.

Game Name
Robopon (HuC-3 mapper)

Analogue OS
v2.0 v2.1

Issues
1. Year and date are saved in the save data, but not the time.
2. Clock information is not loaded from Analogue Pocket at startup. 
In other words, the RTC on HuC-3 is not working.

For example
If the time is saved with the time set to 2/12 16:00 and loaded at 2/13 08:00, the time is set to 2/12 00:00.
The expected result is 2/13 08:00.
Once the clock is set and saved, it will works until the game is finished.

These issues are specific to openfpga-GBC. The original MiSTer works fine.

budude2, 2024-02-16:
Can you send me a save file generated by the mister? Thanks

Lycoris2013, 2024-02-17:
[Robopon_MiSTer_SAVE.zip](https://github.com/budude2/openfgpa-GBC/files/14317511/Robopon_MiSTer_SAVE.zip)
Here it is.
Both US and JP are available.

budude2, 2024-03-07:
I haven’t had a chance to test it, but try it on the latest release. I redid the entire RTC handling

Lycoris2013, 2024-03-08:
I tried those saved data, but the RTC does not seem to carry over.
The time is broken on first load with openfpga-GBC.
It seems to work correctly when reconfigured.

I did a binary comparison and it seems that the data after offset 0x6000 has been changed to different data.
This is data that is not generated by the MiSTer.
At this time, there does not seem to be a one-to-one compatibility with the MiSTer.
If I create new save data, the rtc appears to work correctly. However, the data in 0x6000 is different from the original.

In any case, rtc worked. It is great work.

budude2, 2024-03-08:
Fantastic, that’s great news!

closed 2024-03-08 by budude2