budude2/openfpga-GBC #43: Fast-Forward does not work

DunhamGitHub, 2024-02-27:
If your FF does not work after updating, verify it is enabled

Dawbo66, 2024-02-27:
just checked the 2 cores on mine are both working as they should, but that's with default shoulder button..

DunhamGitHub, 2024-02-27:
> just checked the 2 cores on mine are both working as they should, but that's with default shoulder button..

I again reset everything (FF now set to shoulder buttons) -> does NOT work on my APocket (holding or quick toggling does not go to FF)

_Test cases: digital Roms or Link's Awakening and Pokemon Red DX_

* Previous version of this Core does properly toggle FF

Dawbo66, 2024-02-27:
I just deleted both cores and re applied them, both work for me, so I'm not sure why they won't for you , did you try yesterdays Hallmeier release.? sorry I couldn't help more.

https://github.com/hallmeier/openfgpa-GBC

DunhamGitHub, 2024-02-27:
> I just deleted both cores and re applied them, both work for me, so I'm not sure why they won't for you , did you try yesterdays Hallmeier release.? sorry I couldn't help more.
> 
> https://github.com/hallmeier/openfgpa-GBC

I have not tried the "Hallmeier" release.

I will try to a) delete both cores 1.0.8, b) start the Pocket, c) reinsert the Cores and try again

DunhamGitHub, 2024-02-27:
Installed 1.0.6, FF worked perfectly. 1.0.8 does not work!

Dawbo66, 2024-02-27:
Ok sorry I couldn't be more help. 👍

budude2, 2024-02-28:
> Installed 1.0.6, FF worked perfectly. 1.0.8 does not work!

Did you verify it is enabled in the core settings? It seems to be off by default for some strange reason

closed 2024-02-29 by DunhamGitHub