budude2/openfpga-GBC #47: Original GB DMG filter washed out?

peterlewis, 2024-03-27:
Hi there! :) Has something changed in recent days that might have caused a change in the appearance of the 'Original GB DMG' filter?

I did a `./pupdate` just now and since doing so the colours appear more washed-out than before?

Thanks!

Peter

peterlewis, 2024-03-27:
Ah interesting, it appears to revert to it's previous appearance when setting 'Custom Palette' to 'Off' (as opposed to Auto).

peterlewis, 2024-03-31:
Just for awareness, looks that this issue is being reported more widely, albeit [on Reddit](https://www.reddit.com/r/AnaloguePocket/comments/1bsjrw1/dmg_filter_looks_really_washed_out_even_if_sgb_is/) as opposed to GitHub... :)

Dawbo66, 2024-03-31:
Go into core settings / custom pallet and turn off.. 😁

GiSWiG, 2024-05-13:
As a work around, I've found that playing GB with the GBC core and using the Original DMG LCD video mode the most versatile. Castlevania II looks just like I remembered growing up using the DMG mode but I can switch to the GBC enhancements on the fly. Unless there is a downside I'm missing, using the GBC core for GB games is just great.

budude2, 2024-08-14:
This is fixed in the latest release.

closed 2024-08-14 by budude2