budude2/openfpga-GBC #48: Original GB Core Not Creating Initial Game Saves, But Functions After GBC Core Creates Initial Save Instead

adam1book, 2024-03-31:
I installed the GB and GBC color cores (v1.2.0, latest release) through Pocket Updater (v1.6.0). I copied ROMs from an Everdrive GB X7 into the appropriate folders. When testing saving new games through both the GB and GBC cores so I could tell the save file format to use (since I need to convert the Everdrive saves), I noticed that only the GBC core will save a game for the first time. The GB core will only save games after the GBC core has created the initial save; if the GBC core has not created a game save at least once, the GB core never creates a save folder or file. I tested with both "Kirby's Pinball Land (USA, Europe)" and "Final Fantasy Adventure (World) (Collection of Mana)".

Seems like some kind of permissions issue which the GBC core handles correctly but the GB core does not. I already verified that the newly created save files and the folders they are in are not set to read-only on the microSD card on a Windows computer. Let me know if there's any more info I can provide. I would also appreciate any info you can provide on using Everdrive saves on this core if you have any, hopefully it's a simple conversion.

nlhomme, 2024-04-10:
> I noticed that only the GBC core will save a game for the first time. The GB core will only save games after the GBC core has created the initial save; if the GBC core has not created a game save at least once, the GB core never creates a save folder or file.

Hello, I confirm I am encountering that exact same issue on my Analogue Pocket

TomJinW, 2024-04-26:
I have the exact same problem too.

Oddly enough what I am working on is a custom ROM which requires MBC30 to run. The GB Core can only create 32KiB of saves whereas the GBC Core can create the 64KiB save as it should required by MBC30. After creating the right size of the save file the GB core can save the game correctly.

jakubito, 2024-05-11:
Experiencing the same issue with Tetris - Rosy Retrospection romhack. Creating the initial save with GBC core helped 👍

spikeyxx, 2024-05-24:
Having this too.  GB core cannot save initial saves, but can save correctly if a save is already present. I.e. generated by GBC core.

closed 2024-05-27 by adam1book