budude2/openfpga-GBC #51: Pokemon Red Save deleted

powersurge360, 2024-04-11:
Hey! I'm not sure what happened. I've been playing happily for 10-12 hours on this core and went to play today and it glitched and showed something like a checkerboard pattern. It potentially could have been some other visual glitch, the GBP display filter has some visual noise. That happened twice and on the third boot of the game it showed no active save file. There's still a save file in the right spot that's 30kb, but it seems corrupted. I haven't been able to get Pokemon save editors to read it either. I've tried to include it here but GitHub doesn't seem to want it.

The glitch happened right after the title screen when it should have showed continue, new game, options.

powersurge360, 2024-04-11:
I don't know how reliable it is but I found a [JS save viewer](https://lyndon.codes/assets/html5-pokemon-save-reader/) that I loaded my save into and was able to inspect the JSON blob that came out of it. It looks like a lot of the values are 65533
[image]. This number is interesting because it is (roughly) 256 * 256. I don't know if this is indicative of an error in the core or in the way the save file is presenting data to me.

InfBTra, 2024-04-24:
I experienced a similar case: I was playing Pokemon Red, and after around 5h of game time, without any notable glitch, the save file just suddenly got corrupted. I just turned of the Analogue Pocket yesterday night, turned it ON this morning, and the save file was corrupted.

colloquet, 2024-04-26:
> I experienced a similar case: I was playing Pokemon Red, and after around 5h of game time, without any notable glitch, the save file just suddenly got corrupted. I just turned of the Analogue Pocket yesterday night, turned it ON this morning, and the save file was corrupted.

OMG the same thing happened to me today! I was playing Pokemon red last night, saved in game and turned off the pocket. This morning when I turned it back on the save was gone. Luckily I backed up my save file a few days ago but still lost a good few hours of progress. 😔

SaulGoodman1337, 2024-06-14:
Same here some time ago

jz-k, 2024-08-08:
Not sure if related, but my Pokemon Red file just lost hours of progress after playing on the GBC core with the SGB option enabled. I had just saved when the controls locked up and the game became unresponsive. Water still animated in-game but nothing happened whenever a button was pressed. This includes the Analogue OS button!

I had to hold down the power button for what seemed like a longer time than normal for the system to shut down. I waited a few seconds, restarted, and when I resumed the game it set me back 2 hours to an earlier save. I had saved in-between the resumed save and the save that broke the core, no idea what happened to those in-between saves. I did not get a corruption message but the data is just gone.