budude2/openfpga-GBC #52: Colour change in Pokemon blue

KimiRaiki7, 2024-04-14:
In Pokémon Blue the color often changes briefly during battles (if sgb mode is enabled)
Is this a known problem?

KimiRaiki7, 2024-04-19:
https://github.com/budude2/openfpga-GBC/assets/167031783/0a5a453e-da2e-4cbd-aa0d-05a454746fdc

here you can seethe problem: the graphics change the color during the fight

KimiRaiki7, 2024-06-07:
> Is it happening in other games? Looks like your pocket is breaking...

No, I only noticed the problem in pokemon blue/red. Did you try it on your pocket?

Anatrok, 2024-06-16:
I can confirm this behavior.  If you compare to the spiritualized SGB core (or a physical SGB) this brief color cycle does not occur.

I noticed that on booting the game the boot screen is using a green DMG-esque palette before flashing to black, the back to white with the expected SGB palette.

It seems that during any sprite palette swap (Pokémon red/blue does this on SGB so every Pokémon has an accurate palette) this core has a frame where both sprites default back to the dmg-esque palette.  This is just my laymen's guess.

budude2, 2024-08-20:
That’s odd… Any idea if it does it on mister?