budude2/openfpga-GBC #60: Audio Pop After Sounds Play

Hansukie, 2024-06-17:
I'm not sure if this is present in all games, but for Pokemon Crystal, there's an audio pop after each sound. 

It comes after the gbc splash screen ping, it comes after the Unknown pokemon sounds in the intro cutscene, and when the pokemon pop up out of the brush. 

This audio popping noise isn't present when playing a real crystal cart or when playing on a different gbc openfpga core.

dsrules, 2024-06-26:
I'm experiencing this aswell except on Pokemon Polished Crystal 22c75723, occurs most noticeably when menu related UI sounds are played. Wasn't experiencing this in the previous build. Will test other games.

Hansukie, 2024-06-26:
This also happens in the Zelda Oracle of Seasons/Ages games. 

It happens when browsing the start menu and when walking in puddles of water in caves.

Hansukie, 2024-07-01:
> I'm experiencing this aswell except on Pokemon Polished Crystal 22c75723, occurs most noticeably when menu related UI sounds are played. Wasn't experiencing this in the previous build. Will test other games.

It seems like this is actually normal behavior on actual hardware. Dug out a Game Boy Color and the popping noises happen in Pokemon Crystal. 

I kind of wish there was a toggle for this audio quirk so we can have an option to turn it off. The pops aren't heard in spiritualized's gbc core.

budude2, 2024-08-14:
> It seems like this is actually normal behavior on actual hardware. Dug out a Game Boy Color and the popping noises happen in Pokemon Crystal.
>
> I kind of wish there was a toggle for this audio quirk [...]

I can look into it, but if the popping also occurs in the upstream Mister core I would recommend creating an issue there

MP2E, 2024-09-05:
Looks like the MiSTer core now has an option to silence the pops, though with the warning that some sounds won’t be accurate