budude2/openfpga-GBC #61: Can't seem to get any .gb games to read and play on the openfpga-gb-1.2.2 core.

slockness, 2024-07-04:
Hello, I have the earliest analogue pocket and I can't get the GB core v1.2.2 to work. As soon as I click on a game it takes me back to the game selection and nothing reads. I have the correct bios gb_bios.bin in the folder and all files were copied over correctly on windows 11. One thing to note, the last version that worked for me was v1.1.0 and sometimes the games would display a framework error after a few plays....but it worked for the most part. Also, I have no problem with the gameboy color core....that one works just fine. Not sure why I can't get it to work.

lunchpacked130, 2024-09-10:
[Tl;Dr](www.www.com)

I've sometimes noticed similar strange behaviours, like not loading games properly or taking much longer to load, each and everytime i enter the openfpga menu.

Since many cores are regularly updated (like jotego), as well as some romsets having thousands of small rom files per core, my guess is that the card gets fragmented over time, and in the end it can start causing problems.

Starting fresh with a newly formatted card will put everything you first copy over  "in order", and as long as you never change or delete any files, you can copy files until its full with little to no fragmentation (disorder), and every files location is saved in a file index.

But when files are added, deleted, moved, replaced with never versions etc, it can leave "gaps" in the otherwise ordered files, because a file was deleted, or the replacement file didn't fit where the original file was etc, and those gaps can then be filled up with other files, and other files can be placed randomly around wherever they fit.
So, after a while, the file index gets bloated, and more and more files are "spread around randomly".

Quick formatting the card will erase the file index, and copying all the files over in one go rebuilds it and puts everything "in order".
<details>
<summary><b>Quick Format vs Full Format vs Just deleting files</b></summary>
This is just a simplified explanation of the differences:

• Deleting files: In fact, you don't *actually* delete the files, you just erase their entries in the file index, marking that space as available/free.
So technically the files are still there, and using specialised software and tools, you may be able to recover the files again.
Also, this does not fix any issue with the file index, it's more like using correction fluid instead of an eraser.

• Quick Format: As with deleting files, you're not *actually* deleting any of the files, the main difference is that while deleting files erases just their entries in the file index, Quick Format will erase the entire file index, and create a new clean one.
(As with deleted files, the files are technically still there and might be recoverable with specialised software and tools)

• Full Format: This will fill the entire disk with 0's, and it can do it several times before finishing depending on what settings you use.
Since this operation is overwriting everything on the disk, it is a destructive process to all files that remain, with or without file index entries.
This operation is also very slow and can take uo to several hours to finish.

The reason it overwrites up to several times is to ensure the destruction of any traces of what was on it before, so this is typically only used when you want to be sure that any sensitive/private/etc type files and information on the storage medium is unrecoverable.

(This can also slightly degrade the storage medium if it's a limited write storage like SSDs, USBs and memory cards, as these all have a maximum rated amount of writes they can do before they slowly lose the ability to write or change the files stored.)

For just system, cores, roms and savefiles, and basically everything else you don't care if someone else managed to find, this is overkill and a waste of time, go with Quick Format unless you *really* want to delete everything.

</details>

Also, always remember to properly eject the card everytime you remove it from your computer.
Failure to do this may cause files to become corrupted.

### Tl;Dr:
1. Backup the card
2. Quick Format the card
3. Copy everything back (or re-download everyhing to avoid potentially corrupted files)
4. Remember to eject the card before removing it from your computer.

slockness, 2024-09-10:
Once I did a complete update via the pupdate program, the problem resolved itself...I believe I needed an updated .json file to get it to work since I was doing it all manually in the initial attempt. I love the core!