budude2/openfpga-GBC #66: Playing RomHacks on SD card

Smith-and-einstein, 2024-08-30:
Hi, im looking for playing RomHacks on my SD card with Retropatcher but i can't find the right rom to patch, i only found Pokemon Red ++
Do i have to buy everdrive X5 or everdrive X7 for playing every Pokemon romhacks like Pokemon Prism/ Brown or Super gold 97 etc... ?
Thanks fpr your support.

gLobbob0t, 2024-09-06:
This is not a rom download support page. Please look elsewhere. Btw: retropatcher is different from this core, this is not a issue of this core.

Smith-and-einstein, 2024-09-06:
Im talking about the analog pocket* i forgot to mention it sorry

gLobbob0t, 2024-09-06:
I know that we are talking about the Pocket, but Retropatcher has nothing to do with this core. The patcher is for converting .gb/c rom to .pocket files.
This core only works with .gb/c roms or romhacks. And since you asked about where to get the right rom this is not the place for this.

Please correct me when I misunderstood you.

Smith-and-einstein, 2024-09-06:
No i don't demande rom i demande how it work with .pocket and .gb/c rom with everdrive

gLobbob0t, 2024-09-06:
Doesn't work with .pocket files. This is not the support for Everdrive carts.

.pocket files work with the Pocket natively, you don't need this core for .pocket files.

This is a core. Think of it as a "emulator" for the pocket. With this you don't need a Everdrive or .pocket files. This core can run gb/c rom from the micro sd card. If you want to know how, look at the page for this core.

Btw: you can't demand anything on here. This should be closed since it seems like this has nothing to do with the core.

Smith-and-einstein, 2024-09-06:
Alr thanks you for all

closed 2024-09-06 by Smith-and-einstein