budude2/openfpga-GBC #67: Feature Request - Turbo Button Functionality

lunchpacked130, 2024-09-09:
### ● The Request:
Add turbo button / rapid-fire functionality to the GB and GBC cores.

### ● Why It’s Needed and Some Potential Benefits:
This is inspired by and based loosely on the AGG23 NES core's turbo function.
(You can expand each point for further details/justification)

<details>
<summary><b> Improved Gameplay</b></summary>
• Some games require lots of button mashing, so being able to just hold down a button instead of pressing it continuously as fast as you can, would make those games more enjoyable.
• It can also be useful in other games that doesn't require rapid button presses, for example RPGs, by skipping through text boxes and/or battle menu's for when you're grinding.
</details>
 

<details>
<summary><b> Less Strain / Better Comfort</b></summary>
• Avoiding the need to rapidly press buttons or frequently change your grip can help minimize hand strain. 
• This is particularly beneficial for players who might otherwise adjust their grip to press buttons quickly or to prevent discomfort.
</details>
 

<details>
<summary><b> Reduce Potential Wear</b></summary>
• Fewer button presses can reduce the wear on buttons and contact pads by minimizing the mechanical stress and degradation of these components over time.
• Additionally, it can maintain the responsiveness and accuracy of the buttons for longer, ensuring a better gaming experience.
</details>
 

<details>
<summary><b> Added Accessibility</b></summary>
• Some players may find it difficult or even impossible to press one or more buttons rapidly and/or repeatedly.
• Affected players might therefore avoid playing certain games all together, even if they want to play them, so a turbo button function could let them enjoy those games without having their experience compromised by those limitations.
(Inverted turbo mode could also help for those who find it easier to hold down a button and release it when needed vs having to press it at the right time.)
</details>
 

### ●Additional Turbo Function Requests:
(This will make the Turbo Function even better)

<details>
<summary><b> Option to Enable or Disable Turbo Feature.</b></summary>
(Even though you can have both non-turbo and turbo buttons available, some would like to disable it alltogether to avoid any accidental button presses or other reasons)
</details>
 
 
<details>
<summary><b>Variable Turbo Speed / Button Press Rate.</b></summary>
(Some games would benefit from different turbo rates; f.ex. games that have a limited amount of projectiles that can be on the screen at the same time would benefit from slower rates, spacing the projectiles out more evenely instead of having all of them at once in a burst shot, and having to wait for them to clear the screen)
</details>

<details>
<summary><b> Ability to Toggle Inverted Turbo Mode.</b></summary>
(as the name implies, selected button(s) will continuously rapid-fire. Pressing the button pauses it until released. This can also help improve accessibility (more details in the section about accessibility above)
</details>
 
### ● Example of Function/Layout:
(This is just for extra clarification if needed)

<details>
<summary><b>I've tried here to give a simple example of how it could work in case it's needed.</b></summary>
The GameBoy only uses two face buttons, so on the Pocket, there are two unused face buttons that could be used for turbo functionality, and it makes sense that they are "on the opposite side" of their non-turbo counterparts.

So, if the (A) & (B) face buttons on the Pocket is mapped to (B) & (A) on GameBoy, then (X) & (Y) on the pocket would be (B turbo) & (A turbo) on the GameBoy.

I've added a simple visual representation below.
```
  POCKET          GAMEBOY 
   (Y)              (At)
(X)   (B)        (Bt)   (A)
   (A)               (B)

```
</details>

**Thanks for considering this request!**

budude2, 2024-11-25:
This is already in the core with the right trigger button

lunchpacked130, 2024-12-07:
> This is already in the core with the right trigger button
This is incorrect, my request has not already been implemented.

I think you are confusing my request for `turbo button functionality`, which is **not** implemented in the core, with `fast-forward button functionality` which **is** implemented in the core.

I didn't add "/ Rapid-Fire" to the title of the request, since i figured my request was so carefully explained, that it was redundant, and would become too long..

After all, the first line of my request reads:
`The Request: Add turbo button / rapid-fire functionality to the GB and GBC cores.`

I wish you would read my request and reopen the issue or re-close it with "un-implementable" or something if it's not possible to do, as the reason "already in the core" is incorrect and could potentially be understood by others to mean the core has a functionality that it, atleast at the time of writing this, does not.

Otherwise, excellent work.

budude2, 2024-12-07:
Sorry I had a brain fart. I don’t think this should be too hard to add