budude2/openfpga-GBC #77: GBA Support for RTC?

awolowiecki720, 2024-12-18:
Hey! 
Would love to help any way I can with getting GBA core with RTC support. 

Having the GBC core with RTC has been an absolute game changer for playing rom hacks like Pokemon Crystal Legacy. I and many others would love to do the same for the GBA Pokemon games. 

The go-to core is from [spiritualized1997](https://github.com/spiritualized1997) and hasn't been touched in over 2 years

825i, 2025-02-21:
I would put down some pretty heavy money to have a GBA core even at all, RTC or no. It doesn't look like it's going to happen in the future at all though from what I have seen. We're just stuck with https://github.com/spiritualized1997

nielsderijk, 2025-02-21:
@825i What do you mean? We have had the GBA core from Spiritualized since Summer 2022, right? Plus Budude2 is working on one.

Kurasami, 2025-03-14:
> [@825i](https://github.com/825i) What do you mean? We have had the GBA core from Spiritualized since Summer 2022, right? Plus Budude2 is working on a new GBA core too.

Where is the evidence that he is working on a GBA core? That has been posted absolutely nowhere.

nielsderijk, 2025-03-14:
@Kurasami There's Ko-Fi, Reddit and Discord posts. Even YouTube video's. :)
Have a look here: https://www.reddit.com/r/AnaloguePocket/comments/1hjzrht/new_gba_core_is_coming_along_nicely/

Kurasami, 2025-03-14:
Ok this is _something_ I will admit I haven't seen.

Papaya-Voldemort, 2025-09-02:
Does anyone have any updates on the GBA core with RTC? I have been looking around, and the latest news seems to be around 3 months old, with most of it being 8 months or older. I am about to play Pokemon Unbound but I would love RTC for it as some features use it.

825i, 2026-03-30:
> Does anyone have any updates on the GBA core with RTC? I have been looking around, and the latest news seems to be around 3 months old, with most of it being 8 months or older. I am about to play Pokemon Unbound but I would love RTC for it as some features use it.

Not sure. I gave my Analogue Pocket away but would probably buy another one if a GBA core actually became a thing. Seeing as I got my device mainly just to play GBA games and was pretty disappointed with spiritualized's core.

The video in question above has either been deleted or more likely restricted to only a few people. As is the nature with these things, there's basically no public information. When these things drop they usually do by 100% stealth, which is another reason why I got tired of waiting after about 2 years.

[budude2](https://redlib.catsarch.com/user/budude17/submitted) was making quite a few commits into private repos last year but he seems to have had zero activity this year. So I would say don't hold your breath. The video I saw when it was available showed that it was in more than good enough condition to drop. However there may be anything from legal reasons to anything as to why he dropped off the planet.

EDIT: This was the most recent direct info I could find: https://redlib.catsarch.com/r/AnaloguePocket/comments/1l80c11/budude2_actively_working_on_gba_core_with_ff_were/

tl;dr - He's got life issues (maybe).

He's also got [ko-fi](https://ko-fi.com/budude2) but don't hold your breath.